elementary-robotics/atom #145: dev_atom is not getting created upon composing up docker instances in Macbook

dhruvkakadiya, 2019-04-04:
I am using Macbook with High Sierra OS and I tried to follow steps given in the README file but they are not exactly working for me.

Upto *Building Containers*, everything is working fine but after executing `docker-compose up -d` command, it's creating `atom_atom_1` and `atom_nucleus_1` instances. So I can't run `docker exec -it dev_atom bash` command successfully as there is no `dev_atom` named container.

```
▶ docker container ls
CONTAINER ID        IMAGE               COMMAND                  CREATED             STATUS              PORTS                              NAMES
fe7791a53cfd        atom_atom           "tail -f /dev/null"      55 seconds ago      Up 53 seconds       5900/tcp, 0.0.0.0:6080->6080/tcp   atom_atom_1
25c01d564176        atom_nucleus        "docker-entrypoint.s…"   56 seconds ago      Up 54 seconds       6379/tcp                           atom_nucleus_1

▶ docker exec -it atom_atom_1 bash
root@fe7791a53cfd:/atom# atom-cli
    ___  __________  __  ___   ____  _____
   /   |/_  __/ __ \/  |/  /  / __ \/ ___/
  / /| | / / / / / / /|_/ /  / / / /\__ \ 
 / ___ |/ / / /_/ / /  / /  / /_/ /___/ / 
/_/  |_/_/  \____/_/  /_/   \____//____/  
                                          

                                                                                                                                                                                  
> 
```

So either documentation needs to be updated(if same case is applicable for Ubuntu/Windows OS) or docker script needs to be changed such that it doesn't use some name by default but it stays the same across different OS.

dpipemazo, 2019-04-05:
Thanks for the detailed bug report! You're correct, there's a bug in the documentation, the correct command should be:
```
docker-compose -p dev up -d
```

The `-p dev` part sets the "project name" of the docker-compose session. If the flag isn't passed, it auto-generates the project name from the current folder (atom, in this case). This will almost always be atom, but you'll notice that if you rename the folder in which you cloned the repo your containers would have names that look like `$my_folder_name_atom_1`.

I'll update the docs! Thanks again for the report.

dpipemazo, 2019-04-05:
On a second thought, the more thorough solution is to use the `container_name` field in the `docker-compose.yml` file so that we're not generating container names that may vary based on the version of `docker-compose` we're using. I'll update the PR.

dpipemazo, 2019-04-05:
With the changes in #146 we now set the container name in the `docker-compose.yml` file and are no longer reliant on either the folder name or docker-compose version. You should be able to run the commands as:
```
docker-compose up -d
docker exec -it dev_atom bash
```

dhruvkakadiya, 2019-04-05:
Thank you, @dpipemazo. I tried changes on `145/up_docs` branch and they are working as expected.
```
▶ docker container ls 
CONTAINER ID        IMAGE               COMMAND                  CREATED             STATUS              PORTS                                            NAMES
befb59f0160c        atom_atom           "tail -f /dev/null"      8 seconds ago       Up 7 seconds        5900/tcp, 0.0.0.0:6080->6080/tcp                 dev_atom
32a795404a88        atom_nucleus        "docker-entrypoint.s…"   9 seconds ago       Up 8 seconds        6379/tcp                                         dev_nucleus

▶ docker exec -it dev_atom bash
root@befb59f0160c:/atom# atom-cli 
    ___  __________  __  ___   ____  _____
   /   |/_  __/ __ \/  |/  /  / __ \/ ___/
  / /| | / / / / / / /|_/ /  / / / /\__ \ 
 / ___ |/ / / /_/ / /  / /  / /_/ /___/ / 
/_/  |_/_/  \____/_/  /_/   \____//____/  
                                          

                                                                                                                                                                                                                                                                                                                              
> 
```
So once changes are merged in to the `master`, we can close this issue.

dpipemazo, 2019-04-07:
Great, merged in the changes and will file a new issue for cleaning up the protobuf mentions that are still in the README. thanks!

closed 2019-04-07 by dpipemazo