elementary-robotics/atom #167: Fix bug with how we are using xread by using xreadgroup instead

plipay, 2019-06-25:
Right now, our xread calls in command_send are actually pulling the entire history of messages to that stream, because we are relying on self.response_last_id, which never gets updated.

We should actually move to using xreadgroup, which is a better fit for what we are trying to do here.

mvanveen, 2020-05-19:
### API Changes

**Proposed Process Architecture**

Currently `command_loop` often lives as a separate thread within a parent process.

For example, in `spec-robot` a `CommandThread` is instantiated as an attribute 
of `SpecRobot` and subsequently `start`ed by the parent thread.

The above proposed API change would not change the above portrait at all, but
would instead extend it to fork `n_procs` worth of worker processes.  `n_procs`
would be set `1` by default in the trivial case:

[image]

**Note that in the above scenario, while the parent `command_loop` routine lives in a `CommandThread` it could alternatively live in a separate process after this PR.**

`Element` objects typically behave as singletons AFAIK so multi-process coordination

dpipemazo, 2020-05-20:
@mvanveen this looks great! One thing I noticed:

> the parent command_loop routine would establish a consumer group name for the workers to share, as well as forking out child worker processes

I think this is close but perhaps not quite the desired behavior. It would be quite interesting if the same code, with no modifications, can work whether we spin up 3x copies of an element container with `n_procs=1` or spin up 1 copy of an element container with `n_procs=3`. To do this, I think that there needs to be a globally implied/specified consumer group name for reading from command streams that **all** elements with the same name use. What do you think?

dpipemazo, 2020-05-20:
Also, if we do the above -- I think the multi-processing problem actually became a bunch easier. When we make the children, as long as they have an element **with the same name** as the parent, then we can just so a simple fork/copy and there's no shared memory required. The thread that called `command_loop` would retain the original element, and a deep copy of the element would be passed to each child in `command_loop`. The user of the element needs to ensure that the rest of the app/handlers are process-safe but I think this will simplify it significantly and actually come out quite cleanly.

dpipemazo, 2020-05-20:
In terms of creating the consumer group -- looking at the `XGROUP` docs we can actually just have each individual worker call `XGROUP CREATE` as redis handles if the group already exists cleanly, throws a discernable error, and we can go on our merry command-handling way.

https://redis.io/commands/xgroup

We'll need to come up with a spec to determine how to uniquely make consumer names without conflicting -- something like UUID would likely work.

plipay, 2020-05-20:
This is really exciting, thanks for the detailed writeup!
Couple things I'm interested in the implications of here:

1. Do we need to ensure command/response queue order? Currently, the single-threaded command-loop ensures we always execute/return commands in the order they were received, I think there could be unexpected failures in our system if we don't maintain this guarantee (you don't want to simultaneously end up trying to command the robot to 2 different locations for example, or execute them in the wrong order). What I would like ideally, at least for the default behavior, is to have the command ACKS all returned to their callers in parallel (without waiting for the command to actually get scheduled/executed, we can ack as soon as we receive the command into our queue), but have have the command handlers get called one by one only when the previous one finishes. If we want to support parallelism, thats fine, as long as its a special argument on the command-loop and not the default.

2. There are some tricky implications with `send_command` that we need to keep in mind here as well. If you send out multiple commands on different threads, you need to be able to ensure that you don't "miss" the response for your particular command, which gets processed in the shared command_loop currently. I don't know what multi-threading/multi-processing does to this, but I fixed a nasty bug with this last summer that involved coordination between the two (https://github.com/elementary-robotics/atom/pull/171), and I want to be sure we don't re-break this as part of this redesign.

dpipemazo, 2020-05-20:
Some thoughts:

> Do we need to ensure command/response queue order? 

No, we just won't multi-thread anything that isn't idempotent. By adapting the design to be multi-threaded by nature we give ourselves the ability to use it, but we should only do so where it makes sense. 

> There are some tricky implications with send_command that we need to keep in mind here as well. If you send out multiple commands on different threads, you need to be able to ensure that you don't [...]

Not sure I follow this one -- this is about multi-threading the command-handling side, not the sending side. We're not building out general multi-threading support for sharing the element to send commands from multiple threads. That generally shouldn't be done and is a different issue.

plipay, 2020-05-20:
Agreed on #1, as long as ideally the default is non-multi-threaded so we don't have to retroactively patch all our repos (better to be able to selectively patch repos to ENABLE multi-threading, vs having to patch them to DISABLE it).

On #2, we actually actively depend on this behavior today already. For a concrete example, spec-robot asynchronously kicks off multiple classification commands while its running through the inspection. This succeeds because of the work I did on that linked PR to make send_command thread-safe, its a useful feature we already have which allows us to share the same element instance inside spec-robot and have the multiple threads send commands off that shared instance, and block waiting for their corresponding responses to come in. So basically, I just want to make sure we don't break that default behavior as part of this switch-over.

dpipemazo, 2020-05-20:
> as long as ideally the default is non-multi-threaded so we don't have to retroactively patch all our repos

Yes, good point. `command_loop()` will keep the same behavior. It will take an optional argument, `n_procs` (default to 1) to specify the number of threads. @mvanveen to confirm but I think this is the case

>  we actually actively depend on this behavior today already. For a concrete example, spec-robot asynchronously kicks off multiple classification commands while its running through the inspection

Interesting. I do think it will work as you're describing -- Each command-handling "thread" will actually be its own process so it should be fine. I'm not sure why there's state being shared between:
- a thread that's sending commands (XADD onto other element's command queue and XREAD on current element's response queue) and
- thread that's handling commands (XREAD on current element's command queue and XADD on other element's response queue)

since they don't touch the same streams. There would be potential multi-threading issues between multiple threads sharing the same element's response queue but in that case I would encourage you to just use a unique element for each thing that sends commands since there's no need for it to be the same element name that's serving commands. We really shouldn't need to deal with multi-threading on the same element and should move to a design where each thread/process that needs to send commands/read data spins up its own element. This will also help us trace commands/responses through the system a bit better if each thread has its own element with a unique name.

mvanveen, 2020-05-20:
> Yes, good point. command_loop() will keep the same behavior. It will take an optional argument, n_procs (default to 1) to specify the number of threads. @mvanveen to confirm but I think this is the case

Yep, my current plan was to default `n_procs` = 1. :+1: 

I'm not sure I fully 100% grok the case @plipay is bringing up.  I did see some of that delicate handling in `command_send` previously- in fact this is partly what led me to believe it may relate more directly to this task earlier on in my journey to complete this ticket.

I suppose the concern would come up in a situation where for ex. two consumer threads are performing `XREAD` on the same command response stream in the middle of 2 `command_send`s in parallel, because there would be ambiguity about which command Ack they're receiving on a given channel, right?

I'm aligned with @dpipemazo philosophically that it would be ideal for each thead/process sending commands and receiving acks to have isolation and not have possibility to interfere with one another (in contrast with `command_loop` command consumption, where we're OK with multiple command receivers pulling from a command stream concurrently in cases we can guarantee idempotency and aren't worried about interleaves etc as we discussed above).

I noticed spec robot is currently leveraging a multi-threaded pattern with a `CommandThread` and `RobotThread`.  Is this where something like this would come up currently @plipay?  I was under the impression there was only one command thread active, would this come up if there's multiple live `CommandThreads` that could interleave?

I think for now if it is alright I would like to focus on getting the `command_loop` semantics down sufficient for a V1 which would enable the approach Dan highlighted where all element `command_loop` invocations would leverage the same logical consumer group.

My hope is that we can handle downstream refactoring as a subsequent exercise given the default of `n_procs` = 1 and gradually increase concurrency through the system.

Thoughts?  Apologies for any confusions in the above- I'm still learning and grokking the specifics of the command protocol so please excuse naivete or possible misunderstandings.

dpipemazo, 2020-05-20:
> I think for now if it is alright I would like to focus on getting the command_loop semantics down sufficient for a V1 which would enable the approach Dan highlighted where all element command_loop invocations would leverage the same logical consumer group.

Seems like a good plan to me. I think this is a great place to start and I am pretty sure it will all be OK downstream with the current use cases inside of ER.