elementary-robotics/atom #182: Better documentation of xhost permissions

plipay, 2019-08-07:
`xhost +SI:localuser:root` is sometimes required before you can run GUI atom apps, should be better documented

Cody-G, 2019-09-18:
I can make this clearer in the docs, but currently executing that command isn't enough for me to get everything running locally.  I haven't yet found a way to get all of the GUI apps from the [tutorial](http://robotics.atomdocs.io/#getting-started-with-atom-os) to work locally, despite using that command and trying several modifications to the docker-compose file.  I did find one configuration that worked for everything except the simulator app.  Can anyone else run everything locally at the moment?

Cody-G, 2019-11-13:
This was mentioned a few times in the walkthrough, but it wasn't so easy to find.  I've added it to a new SDK-Graphics section of the docs in #224

PR #225 (2019-11-15): Clarify X11 graphics with and without GPU in docs
Closes #182.
closed 2019-11-15 by dpipemazo