elementary-robotics/atom #184: Decrease default command/response stream length

dpipemazo, 2019-08-08:
I believe across C, C++ and python the default command/response stream length is 1024 entries which wastes a bunch of memory when we're passing around camera images. Shorten this to something reasonable like 100. This is tied to #183 since we're hitting our max mem limit until this is shortened. @plipay thoughts on appropriate length?

Cody-G, 2019-09-18:
I see where to make these changes, just waiting on a suggestion for the new length.  I can do 128 since everybody loves powers of 2?

Cody-G, 2019-11-07:
@dpipemazo in light of the new reference API do we want to reconsider this?

dpipemazo, 2019-11-08:
@Cody-G agreed. With the new reference API we can close this.

closed 2019-11-08 by dpipemazo