elementary-robotics/atom #192: Fix error of usage of entry_read_since in docs

cheripai, 2019-09-05:
The docs state that
```If last_id is not passed, this call will return the first new piece of data that's been written after our call.```
However, this is not true as if `last_id` is not passed, we return the first piece of data in redis.

If we want to get the latest piece of data since the call, we should set `last_id = element._get_redis_timestamp()`

dpipemazo, 2019-09-05:
@cheripai that might work but it's not the correct usage, we should be calling `XREAD BLOCK ...` with `$` as the correct way to do this. It's natively supported in the `XREAD`, the `$` says to wait until the next piece of data is published.

https://redis.io/commands/xread

Cody-G, 2019-09-18:
I see how to implement the blocking behavior that @dpipemazo mentioned, but there does still seem to be a conflict in the docs that will need to be resolved.

[Here](https://github.com/elementary-robotics/atom/blob/349276d3a50c87107916ee685bf58e2fb26be58d/languages/python/atom/element.py#L631) we set `last_id` to "0" by default, which as documented [here](https://github.com/elementary-robotics/atom/blob/349276d3a50c87107916ee685bf58e2fb26be58d/languages/python/atom/element.py#L641) means the default behavior is to return all entries (ordered by recency).  That conflicts with what @cheripai quoted from the tutorial, `If last_id is not passed, this call will return the first new piece of data that's been written after our call`.

One way to resolve this is to set the default value of `last_id` to `None`, and interpret this as "block until you've read `n` new samples after the call".  That's a breaking change if people have been relying on the default value of `last_id`.

Or alternatively we could just change the tutorial, and I'll add code to handle the case `last_id is None`.  That shouldn't break anything.

PR #194 (2019-09-25): Modify and test the default behavior of entry_read_since (reads new e…
…ntries only).  Fixes #192.
closed 2019-09-25 by dpipemazo