elementary-robotics/atom #195: Development within an atom container

Cody-G, 2019-09-19:
Per the README:

>It's best to do all developmental testing and building within the docker containers. The easiest way to do this is to mount the current directory into the latest atom image that's been built and continue development in there until you're ready to rebuild.

I think I like this idea, but the current atom docker image doesn't have a text editor installed for development.  Maybe we could install an editor or two in the Dockerfile?  Or is there another way to do this?  I'm new to docker and docker-compose so I may be missing something.

dpipemazo, 2019-09-20:
@Cody-G this section mainly covers how to link your files into the container s.t. you can do your text editing on your host but have the files mounted into the container so you can test them :smiley:

Cody-G, 2019-09-20:
Ah I see, thanks!  I found exactly how to mount a local folder to `/development` in a different section of the docs (pasted below)

```
  $element:
    container_name: $element
    image: $image
    volumes:
      - type: volume
        source: shared
        target: /shared
        volume:
          nocopy: true
      - ".:/development"
    depends_on:
      - "nucleus"
```
This seems like a great way to develop a new element, but now I'm wondering what's the best practice for developing atom itself?  I've been working on atom's python api and getting by without building new images all the time, but if I were making core changes to atom then I think I would need to rebuild the image often?

dpipemazo, 2019-09-30:
This is a good point -- typically I link the source for atom into the container and then call the uninstall/reinstall commands for the language client directly. It's definitely more tedious. Going to close this issue though.

closed 2019-09-30 by dpipemazo