elementary-robotics/atom #197: RFC: Julia language client

Cody-G, 2019-09-24:
I now have a draft of a julia language client in a branch at https://github.com/elementary-robotics/atom/tree/cjg/julia_clang.  At the moment I'm wrapping the atom C client, so the interface is exactly the same as the C interface.  I see two big advantages to wrapping in this way: that we get the high performance of the C client (Julia can interop with C without overhead) and the wrapper can be regenerated fairly easily when the C code changes, meaning we have less code to maintain.

The disadvantage is that at the moment the C api seems to be missing features present in the Python api.  For example, I don't see any support for pipelining.  Probably my favorite way to handle this would be to add features to the C client to get it in line with the Python client, and then just regenerate the Julia wrapper.  Alternatively we could just abandon the wrapper idea and write a new client from scratch in Julia.  Or take an intermediate approach where we implement only the missing features in Julia, but that will probably be too messy.  In any case it will be a time investment to get to feature-parity with Python, so I'll hold off on this now until we understand whether it's worth it.

dpipemazo, 2019-09-30:
@Cody-G Let's get a PR up for this. It would indeed be nice to get pipelining into the C API (it's supported by `hiredis`, the underlying redis client), but we don't pipeline that much overall. Most of our interactions with redis are single-command.

dpipemazo, 2019-09-30:
Please update the docs with a `julia` tab for explaining how to make the calls in julia when getting the PR up