elementary-robotics/atom #201: Move Docker container base from ubuntu to `python3.6-slim`

dpipemazo, 2019-10-01:
The container is pretty bloated -- we should look into making it much slimmer in order to speed up builds and minimize download sizes.

aywaldron, 2020-01-09:
@dpipemazo it looks like `python:3.6-slim` (174MB uncompressed on-disk) is not actually smaller than `ubuntu:18.04` (64.2MB). I was able to get the final image size down to 1.04 GB from 1.1 GB by removing a couple things but not sure what else to try for reducing size (assuming you don't want to move to alpine).

dpipemazo, 2020-01-09:
@aywaldron this is interesting! Nice investigation. Alpine would be nice if the VNC would still work -- there's nothing really keeping us on `ubuntu` I think.

aywaldron, 2020-01-09:
One concern with alpine is that it uses the `musl` C library instead of `glibc`, which has been known to cause some incompatibilities (see [this article](https://pythonspeed.com/articles/base-image-python-docker-images/) for some examples). I'm not sure whether it would cause any issues or not but also not sure if it's worth risking it. Most of the image size right now is coming from all of the packages being installed so I don't know that it would make enough of a difference to justify the risk.

dpipemazo, 2020-01-09:
:+1: sounds like a good analysis -- we'll hold off then. Closing this issue.

closed 2020-01-09 by dpipemazo