elementary-robotics/atom #210: Add `reference` API to Atom

dpipemazo, 2019-10-31:
We currently have an issue when we're passing large images back and forth within atom. Each time we pass an image as an argument to a command, it gets put onto the command stream for an element. Command streams are default 1000 entries long -- meaning an element which is primarily used for image-based commands can have up to 1000 images sitting in its command stream. The same happens when we're passing images as responses.

As such we can waste a bunch of memory (gigabytes, even). It also makes the command/response logs harder to view/parse in `atom-cli` which is unfortunate.

A solution for this would be a concept of a "reference" in atom. A user would be able to generate a reference for an arbitrary piece of data. The data may or may not be msgpack'd. Under the hood, a reference would be an expiring cache entry in `redis` with a default timeout of something like 60s. The timeout would be settable.

The new sequence for calling an image-based API would be:
1. Caller: Call `create_reference(data=image, msgpack=False, expire=10)` to create a reference for an image that will time out in 10 seconds here. This will return some unique string. Under the hood it could call a redis `SET` command.
2. Caller: Pass the unique string returned by `create_reference` in the command data payload instead of the image itself
3. Responder: After getting into the command callback, call `get_reference` on the unique string to receive the data itself from redis. Under the hood it could call a redis `GET` command.
4. Responder: Continue as normal
5. Caller: Upon receiving the response, and when no longer in using the data for any other commands, call `delete_reference` on the reference. This will call DEL/UNLINK on the reference and clean it up. If this step is skipped (as will eventually happen), the reference will expire after the timeout anyway.

Similar process for returning an image: the responder creates the reference and returns it instead of the data. The caller then is responsible for cleaning it up when it's done with it.

The final thing to be hashed out above would be how we make our keys for references. Our keys for references need to be unique, otherwise we're going to have reference collisions and corrupted data. They can perhaps begin with `reference:$element_name:` and then have some UUID tacked onto them. That still runs some amount of risk of collision, but it's pretty minimal. We could use the `INCR` command to have a global reference count in the system but then we're adding another redis transaction to the mix which might slow things down. I think I'm OK with `reference:$element_name` and perhaps a nanosecond-level timestamp as it's highly unlikely we have two references created in the same element at the same nanosecond. Interested to hear everyone's thoughts on this.

Overall I think this will be a nice feature and will:
1. Save us GB of memory
2. Make the logs easier to follow
3. Perhaps save us some copy overhead if we're sending the same image back and forth a bunch of times as we are in the system.

dpipemazo, 2019-11-09:
@cheripai has merged in support in Python, still need C++

Cody-G, 2019-11-15:
While updating the image alignment element to use references I noticed that `reference_create` returns a list while `reference_get` returns a dictionary keyed by reference id.  This was unexpected for me, and will be a bit hard to remember.  It's also more verbose in practice.  Is anyone in favor of changing `reference_get` to return a list?  The code below demonstrates the current behavior and the proposed behavior.  If people like the proposal I'm happy to implement.

```python
import atom
import numpy as np
import cv2

element = atom.Element("demo")

#create two images that we'll reference
img0 = np.zeros((100,100))
_, img0_serialized = cv2.imencode(".tif", img0)
img0_bytes = img0_serialized.tobytes()

img1 = np.zeros((100,100))
_, img1_serialized = cv2.imencode(".tif", img0)
img1_bytes = img1_serialized.tobytes()

img_list = [img0_bytes, img1_bytes]

#create a list of references from a list of images
ref_list = element.reference_create(*img_list)

#current way to get a list of images back from Redis:
img_dict = element.reference_get(*ref_list)
img_list_got = [img_dict[k] for k in ref_list]

#after proposed change:
#img_list_got = element.reference_get(*ref_list)

```

dpipemazo, 2019-11-16:
@Cody-G i think the feedback makes sense. I did think about this a decent amount when implementing/creating the spec. The use case that I optimized for was “I want to get a list of references and then address a specific one of them”. In this use case I think the dictionary is the right return type — I don’t find remembering and using indices into a list to be a particularly great way of keeping track of which blob of binary data to use. It forces the user of the data returned by ‘reference_get’ to know the context of the original al list to interpret the data which I don’t particularly like. With the way it’s implemented, one can simply look at the dictionary itself to figure out how to interpret it — imagine if this list were passed to a function, I think you’d also need to pass the original list of references if you wanted to grab a particular image. 

That said, it is a tradeoff, but I question why you need to convert it into a list at all — why not just iterate over the dictionary? Ordering might matter which would be a good point

Cody-G, 2019-11-16:
> I don’t find remembering and using indices into a list to be a particularly great way of keeping track of which blob of binary data to use. 

I do agree with that, but `reference_create` already demands that we keep track of which reference corresponds to which piece of data passed to `reference_create`, and I don't see an easy way to have `reference_create` return a dictionary -- unless we key the response by the piece of data itself, which seems clunky.  I'm not so much disputing whether the return type should be a dictionary or a list, just suggesting that `reference_create` and `reference_get` use the same return conventions.

> imagine if this list were passed to a function, I think you’d also need to pass the original list of references if you wanted to grab a particular image.

I think if you're passing a list of (de-referenced) data to a function then it may be dangerous to rely on the reference key as a persistent identifier since references are ephemeral by design.  That said, I can imagine some use cases where having an O(1) mapping from reference key to data could be useful.  So far in my applications I haven't made use of this because raw reference ids are hard to interpret without referring back to the data passed to `reference_create`, which is a list.

>That said, it is a tradeoff, but I question why you need to convert it into a list at all — why not just iterate over the dictionary?

By making `reference_create` depend on the order of arguments I think we set the expectation that order matters. If the user wants an O(1) mapping from the pre-`reference_create` data to the reference id then they first need to manually create a dictionary in O(N).  I don't see a real need to use lists over dicts in general, and I agree that a dictionary is more flexible -- I just think mixing lists and dicts is unintuitive in this case.

dpipemazo, 2019-11-19:
Nice points @Cody-G, as implemented in #229 the standard is now to return a list in which the order matches the arguments to the function.