elementary-robotics/atom #211: RFC: Add/create `docker-compose` configurator per element

dpipemazo, 2019-11-06:
We currently have a pain point in `atom` where it's tricky to figure out how to create a `docker-compose` config for an element. If you want to run an app with multiple elements you need to consult the READMEs for each element, copy the `docker-compose` configuration and string it all together manually. You also need to hunt through the docs for mac/linux/windows config options and insert them into every docker-compose file correctly.

At the end of the day, a user should just be able to specify:
1. Which elements to run/
2. Which OS they're on
3. Any per-element config options in simple variables (not necessarily the names of the ENV variables)

And we should be able to spit out a `docker-compose` file with the correct configs. This would also help so that when we update the element and are running a new version we pick up/update the docker-compose config.

I would propose as a starting point for discussion the following:

1. We make a tool in Atom, `AtomConfig` or similar, that's perhaps a Python library and base class.
2. The base class will have standard graphics/USB/tmpfs/network options that will be shared amongst all elements. 
3. The base class will have a way of extending its configuration for custom env vars, file mounts, etc.
4. The base class will have a `to_docker_compose` function that will generate the YAML for a docker-compose file. This generality allows us to eventually extend more abstracted config settings to other systems like k8s.
5. We ship `AtomConfig` in the base Atom docker container and expect each element to implement a class that extends it/defines its own config. This file should be a Python file at the top-level of the element that shares a common name like `config.py` and *must* be included in the docker container shipped for the element.
6. In order to keep the version of the element and its config in sync we can use Docker multi-stage builds where we can use COPY FROM some-element-image to get the proper config class for the version of the image we're running. Since we need to pull the containers to run the app anyway, there's no additional overhead here.
7. We make a tool called `atomize` or similar that takes a list of elements and a single `.ini` file or similar and will run the process in (6) and spit out a single docker-compose for the whole system.

This will make keeping configurations fresh a bunch easier and ease some of the `docker-compose` pains some of the developers see. It will also make it easier to switch platforms from `docker-compose` in the future as now the developers only need to understand the configuration options and don't need to understand much about the system we're using to run the docker containers. Particularly this will help with a shift to k8s if/when that happens.

This will also improve reliability on deployment as it won't take PRs on elements + repos that keep track of deployment configs when we update/change docker-compose dependencies.

Cody-G, 2019-11-06:
Just to clarify for myself, would the workflow look like this?

1. Developer writes a new element, element-c that requires element-a and element-b to run
2. Rather than hand-writing a docker-compose yml file for element-c and committing it to the element-c repo, the file is generated __at deploy time__ by sending the `atomize` tool a list of elements and config variables that may be specific to deployment conditions.

If that's correct, then I suppose we wouldn't check in any docker-compose.yml files anymore?  For example, we'd add a Circle build step that would call `atomize` to generate the docker-compose yml dynamically?

If that's all correct then I think my main question is what the interface to `atomize` looks like.

dpipemazo, 2019-11-08:
@Cody-G I think this is more for an end-user of elements who wants to run an "app" consisting of A, B and C elements. The user is on a particular OS and wants to customize a few things.

They pass the elements they want to use, their host OS, and configurations to `atomize` which will then spit out a docker-compose file that will launch the app and is tailored to their liking.

Creators of elements can also use `atomize` in order to develop their own `docker-compose` files for their repos, but in general they'll probably commit their docker-compose files that have been tailored to their development needs to their repos as before.

A suggestion of an interface might be a config file (any markup language, can choose one) that has a section per element that wants to be used along with any customized parameters for that element. This is passed to `atomize` to generate a `docker-compose` file.

dpipemazo, 2019-11-14:
Thinking about this a bit more -- I do see a need for `atomize` in that it's a bit limiting to only support `docker-compose` with Atom. As we move forward we're going to want to be able to easily generate kubernetes, Amazon ECS, etc. configurations for containers to make the same containers easy to configure and deploy for single-machine or multi-machine deployments. `atomize` would fill a need here to take the elements and allow users to easily string them together for a bunch of different runtime configurations.

dpipemazo, 2020-01-24:
We're going to hold off on it for now -- currently handling internally with `ansible` and its jinja2 templating. Would ideally be able to expose these templates within atom and we can decide how best to do that down the road.

closed 2020-01-24 by dpipemazo