elementary-robotics/atom #230: RFC: Integrate Apache Arrow as a serialization option

dpipemazo, 2019-11-19:
@Cody-G has pointed out a few use cases where our current serialization options aren't optimal -- specifically large, structured arrays/matrices that aren't necessarily images. In these instances `msgpack` is [not great](http://satoru.rocks/2018/08/fastest-way-to-serialize-array/) and [Apache Arrow](https://arrow.apache.org/) is much better. Arrow's language support, though not at the same level as Redis's or MsgPack's is quite solid and would likely cover the majority of use cases in which one would be processing such arrays.

Interested to hear thoughts on:
1. Adding Arrow as a serialization option
2. If (1), then should we change the serialization args from T/F to one of a list of options, such as in Python, changing from:
```
entry_write(..., serialize=True)
```
to
```
entry_write(..., serialization="msgpack")
```
or similar

Cody-G, 2019-11-19:
I only just found out about Apache Arrow, but this sounds great!  It should be faster than msgpack by a few orders of magnitude, and in many cases numpy arrays can be decoded from arrow arrays without copying memory.

As for the implementation, I prefer to make `serialization` a string keyword argument as @dpipemazo suggested.  If people are concerned about disruption then a less breaking change would be to keep `serialize` as a T/F switch, and then add _another_ keyword argument `serialization` that sets the default serialization backend to msgpack, and that argument would be ignored if `serialize` is False.

dpipemazo, 2019-11-19:
I like the idea of keeping `serialize=True` for legacy purposes and then adding the new option :+1:

aywaldron, 2020-01-10:
@dpipemazo Does this issue pertain to just the `entry_write` method or other methods that have the `serialize` option as well?

dpipemazo, 2020-01-10:
@aywaldron this would apply to all APIs. It would allow the user to select a different option from `msgpack` when choosing to serialize leaving their choices as:
1. No serialization (raw data being sent/received)
2. Serialization with msgpack
3. Serialization with arrow

It would also need to be added into `atom-cli` so that we can send/receive/read arrow-serialized data

PR #234 (2020-02-03): Issue #230 - Integrate Apache Arrow serialization
- Adds str `serialization` option to Element and Response API; defaults to `msgpack`
- Changes boolean `msgpack` CLI setting to str `serialize`; defaults to `msgpack`
- Adds Apache Arrow as serialization option; installs `pyarrow` in atom and nucleus docker images
- Pulls serialization functionality into new `atom.serialization` module
- Adds `"ser"` key for serialization method to entries; uses its value to deserialize entries on read
- Adds serialization information to reference keys and uses for deserialization of references
- Adds serializa [... cut]
closed 2020-02-03 by aywaldron