elementary-robotics/atom #239: Enable git-lfs installing in org

Cody-G, 2020-01-27:
crossref https://github.com/elementary-robotics/element-barcode/pull/7#discussion_r371334802

It would be great to alter the Atom orb to make it easier to install and use git-lfs.  Below are the steps I took to install git-lfs in the CircleCI config of element-barcode, which also required setting `executor: atom/build-ubuntu`.

```
        - run:                                                                                                          
            name: Install Git Large File Storage (LFS)                                                                  
            command: |                                                                                                  
              curl -s https://packagecloud.io/install/repositories/github/git-lfs/script.deb.sh | sudo bash             
              sudo apt-get install -f git-lfs && git lfs install                                                        
              mkdir -p ~/.ssh/ && echo -e "Host github.com\n\tStrictHostKeyChecking no\n" > ~/.ssh/config               
              ssh-keyscan -H github.com >> ~/.ssh/known_hosts                                                           
              ssh git@github.com git-lfs-authenticate "${CIRCLE_PROJECT_USERNAME}/${CIRCLE_PROJECT_REPONAME}" download  
```

aywaldron, 2020-01-30:
@Cody-G @dpipemazo A development orb tagged `elementaryrobotics/atom@dev:239` has been published to test the addition of this command. Since we recently merged in other changes to the orb, testing with this dev orb can also test that those changes are not breaking existing builds. Let me know if you want me to try it out on `element-barcode`.

Cody-G, 2020-01-30:
>Let me know if you want me to try it out on element-barcode

That would be great!  So after your changes can we eliminate all of [these](https://github.com/elementary-robotics/element-barcode/blob/master/.circleci/config.yml#L19-L26) lines?  Or we still need to authenticate?

aywaldron, 2020-01-30:
@Cody-G I included the authentication in the install command so yes. We can also split them out into separate commands if we want the flexibility to not have to run installation and authentication at the same time.

Cody-G, 2020-01-30:
I don't see a problem with running installation and authentication at the same time, as long as the authentication in the orb is run again again at the time of the job (don't know enough about how orbs work to answer that myself)

aywaldron, 2020-01-30:
It will be run on every job. 
I created a ticket at https://github.com/elementary-robotics/element-barcode/issues/10 for updating the `element-barcode` build.

Cody-G, 2020-01-31:
Woohoo, closed by #242

closed 2020-01-31 by Cody-G