elementary-robotics/atom #251: Decrease build times

aywaldron, 2020-02-06:
- Move `arrow` build to `atom-base` image.
- Remove `msgpack-c` build from `nucleus` image.
  - Clean up `nucleus` image to reduce size.

dpipemazo, 2020-02-06:
Other things to do:
1. Remove "base" dockerfile from separate build and take advantage of DLC in circle now that we have it again
2. Set machine size to `2xl` in executor to bump from 2 cores to 8 cores

dpipemazo, 2020-02-06:
If possible, in the executor, can we parameterize the size of the instance to be used? for compute-intensive builds we're going to want to use `2xl` but for uploads we're going to want to use `small` as they get billed differently.

dpipemazo, 2020-02-06:
https://circleci.com/docs/2.0/credits/

Cody-G, 2020-02-06:
@aywaldron for what it's worth I just pushed my work toward shrinking the base and main atom images:
https://github.com/elementary-robotics/atom/pull/253

With those changes I was getting a 730MB base image and 1.18GB main image (when building locally, not when compressed on dockerhub).  I opened a PR just to see if the tests pass.  I have a feeling the C++ tests won't pass since I removed some debugging tools that may get used in the tests, but we could fix that by installing them at test-time rather than baking into the image.  I'll stop working on this now because I don't want to duplicate your work, but let me know if you want to collaborate on this

aywaldron, 2020-02-06:
@dpipemazo The `resource_class` is set on a per-job basis so not necessary to parameterize since it can just be specified on each job, e.g.: 
```
jobs:
  build-atom:
    executor: atom/build-classic
    resource_class: large
```
One potential issue is that the `machine` executor [only allows for medium+ resource classes](https://circleci.com/docs/2.0/configuration-reference/#machine-executor-linux), not small. We might want to consider switching executors for jobs like deploying to dockerhub in order to use a small resource class.

aywaldron, 2020-02-06:
Important caveat for Docker Layer Caching from [docs](https://circleci.com/docs/2.0/docker-layer-caching/):

> DLC caches your Docker image layers by creating an external volume and attaching it to the instances that execute the machine and Remote Docker jobs. The volume is attached in a way that makes Docker save the image layers on the attached volume. When the job finishes, the volume is disconnected and re-used in a future job. This means that the layers downloaded in a previous job with DLC will be available in the next job that uses the same DLC volume.
> 
> One DLC volume can only be attached to one machine or Remote Docker job at a time. If one DLC volume exists but two jobs that request DLC are launched, CircleCI will create a new DLC volume and attach it to the second job. From that point on the project will have two DLC volumes associated with it. This applies to parallel jobs as well: if two machine jobs are run in parallel, they will get different DLC volumes.
> 
> **Depending on which jobs the volumes are used in, they might end up with different layers saved on them. The volumes that are used less frequently might have older layers saved on them.**
> 
> The DLC volumes are deleted after 7 days of not being used in a job.

aywaldron, 2020-02-07:
In trying to optimize DLC we ran into an issue with installing arrow without the rest of the atom repo (which cannot be added before building arrow if we want to cache the arrow build layer). To fix this, instead of tracking down the arrow install issue, we are moving the nucleus image to `python-slim` so we can install pyarrow with pip rather than build from source and save significant build time.

PR #254 (2020-02-11): Issue #251 - Update build for new plan
- Remove `--no-cache` option from `docker-compose build` command
- Increase build machine sizes to large

Published dev orb ` elementaryrobotics/atom@dev:251`

Resolves #251
closed 2020-02-11 by aywaldron