elementary-robotics/atom #289: Ignore W503 in flake8 config

Cody-G, 2020-03-16:
Per the [flake8 docs](https://www.flake8rules.com/rules/W503.html), the W503 check will go obsolete soon and the opposite behavior will be encouraged (line break before operator will be the default).  Currently our flake8 config throws an error for W503.  I propose we ignore this error by adding this to the config:

```
ignore = W503
```

dpipemazo, 2020-03-16:
@Cody-G we can discuss, but style changes need to be company-wide. I'm hesitant to adjust it now that it's finally up and running. We have code that conforms to this now.

Cody-G, 2020-03-16:
My first post wasn't clear about this, but the issue is that right now our flake8 settings forbid line continuation _both_ before and after an operator, so we can't do line continuation at all.  The only reason this hasn't come up is that we don't have these line continuations in atom.

This triggers a W504 error with our current settings:
```python
a = (3 +                                                                                                                
     4)
```

This triggers a W503 error with our current settings:
```python
a = (3                                                                                                                  
     + 4) 
```

The recommendation from the discussion that I linked above seems to be to tell flake8 to ignore one or the other of W503 or W504 because they're mutually exclusive.  It seems best to ignore W503 because it's outdated based on PEP8.

dpipemazo, 2020-03-16:
@Cody-G the source code seems to ignore W503 as written:

```
          command: |
            echo "[flake8]" > .flake8
            echo "exclude = << parameters.exclude >>" >> .flake8
            echo "max-line-length = 120" >> .flake8
            echo "max-doc-length = 80" >> .flake8
            echo "select = C,E,F,W,B" >> .flake8
            echo "ignore = E203,W503" >> .flake8
            cat .flake8
```

Are we missing something?

dpipemazo, 2020-03-16:
source: https://github.com/elementary-robotics/atom/blob/master/.circleci/atom.yml#L263

Cody-G, 2020-03-16:
Oh duh, you're totally right.  I copy/pasted it wrong from the circle config file to my local machine.  Not an issue after all :-)

closed 2020-03-16 by Cody-G