elementary-robotics/atom #331: Parameterize Element Creation timeout

plipay, 2020-05-22:
Currently, if you try to create an Element instance to connect to some remote host, but the remote host is down, we block for an absurdly long time before raising an Exception. You can test this locally by doing something like:
```
from atom import Element
a = Element('test', host='banana')
```

Most likely, this is happening somewhere in the redis client constructor.
What we would like, is to have this timeout be parameterizable.
Find out what the default is today, and maintain that, but allow us to do something like:
```
a = Element('test', host='banana', timeout_ms=3000)
```
And have the element throw an exception after 3 seconds.

nceruchalu, 2020-05-22:
This will be handy in the heartbeat logic. We currently do a bunch of "interesting" things to work around this.

mvanveen, 2020-05-22:
@plipay any preferences on a default timeout in this context?

mvanveen, 2020-05-26:
:ship: 'd

closed 2020-05-26 by mvanveen