elementary-robotics/atom #337: Implement Client/Server Elements

mvanveen, 2020-05-28:
# Client / Server `Element` Design Proposal

## Introduction

Client/Server element use cases have become more common within Atom.

Providing first-class support for this pattern with `ClientElement` and
`ServerElement` objects can help guarantee proper enforcement of
several properties we'd like to guarantee for client/server interactions
on top of Atom.

## Design Goals

Specifically, we'd like to guarantee:

- Each `Element` performing redis activity should only live on one process (with the exception of multi-process `command_loop` execution)

From this base principle are two subsequent properties we'd like to guarantee for clients and servers respectively:

- **highlander singleton** There can only be one server `Element` of a given name instantiated at a given time.  Subsequent instantiations will fail or exit upon determination that there is an active `Element` of the same name already instantiated.

- **individually addressable clients** While there can be more than one client  `Element` of a specific name.
  
The above goals also have the added benefit that each command stream for a given instantiation of an `Element` is uniquely addressable.  For example if we wanted to inspect a specific client command stream, we could do something like the following due to this granular namespacing:

```
> XINFO STREAM camera:client:b9940b2c-15e5-441b-bc6e-dc3a811d21a2'

```

## Proposed Solution

### Proposed Object Oriented Design

[image]

Both `ClientElement` and `ServerElement` will derive directly from `Element`.  The only difference with respect to these new children classes will be further enforcement and refinement of the `Element` naming scheme, a

### Proposed Implementation Specifics

- Pull request #336 implements parent pid-level `__del__` semantics.  In other words, only a parent `command_loop`ing or instantiating process can clean up streams with `__del__`
- In this ticket we'll take the idea further to provide enforcement that only one `Element` name can be instantiated at a given time.
  - If a subsequent `Element` name is instantiated when a command stream of the same name already exists the instantiation will fail.
  - this can be handled by either checking to see if a given command stream has already been created and ensuring that streams are always cleaned up during creation, or alternatively, via a separate redis semaphore lock, e.g. the pyredis `Lock` object.

**highlander singleton server**

- is effectively a singleton
- cannot live across multiple processes

`ServerElement`'s implementation may not diverge that heavily from `Element`, but its usage
will denote that a particular `Element` is intending to be a singleton server object.

```python
>>> server = ServerElement('camera')
>>> server2 = ServerElement('camera')
RuntimeError: there can only be one 'camera' ServerEleement, and one is already running on pid 1337
>>>
>>> server.name
'camera:server'
```

  
- Unlike `ServerElement`, multiple `ClientElement`s of the same `name` constructor argument can be created
- However, each client will get a unique identifier that is included in the final `Element.name` that 
  appears in redis:

```python
>>> client = ClientElement('camera')
>>> client.name
'camera:client:011cc323-e893-4de1-b08d-6c1812518847'
````

A classmethod on the `ClientElement` class could provide an accessor to 
get all currently active client `Element` names:

```python
>>> ClientElement.get_all_client_names('camera')
['camera:client:011cc323-e893-4de1-b08d-6c1812518847']
>>>
```

dpipemazo, 2020-05-28:
This makes a lot of sense to me and I think will significantly improve the usability of Atom.

nceruchalu, 2020-05-29:
This issue description is nicely detailed and fun to read 😆.
Also this is a cool feature!