elementary-robotics/atom #344: Allow specifying overriding the Redis timestamp when writing a stream entry

Cody-G, 2020-07-20:
I think this is worth discussing.  The main use case is for hardware: sometimes we have a hardware timestamp that is more accurate and relevant than the timestamp automatically generated when writing to a Redis stream.  For example, currently the Basler element calculates a Unix timestamp corresponding to the moment that camera exposure begins.  Instead of writing this timestamp as a field on the stream entry (current behavior) we could use this as the timestamp for the Redis entry.  This will make it easier and more efficient to wait for a camera frame that arrives after a particular time:  Instead of reading each entry and checking the timestamp key (current behavior) we'll only need to read one entry since Redis does the work.

This could be implemented with a new keyword argument to `entry_write` that allows us to override the Redis timestamp.

dpipemazo, 2020-07-20:
+1 on this feature -- it will unlock quite a bit for us.

Cody-G, 2020-09-03:
I looked into implementing this feature.  Some thoughts:

- The main usage of this feature is synchronization of multiple processes/microservices with shared access to a Redis stream.  If we want that to work correctly then stream "producer" and "consumer" processes need to agree on how these timestamps are defined, and they need a common reference.  Example use case: spec-robot wants to query the first image from an image stream that is timestamped _after_ spec-robot's notion of `now`.  Then spec-robot can do an `XREAD` with `now` as the last id.  In order to make sure all of our processes on a machine use the same reference point for `now`, we can use the raw monotonic system time (available in python through the [`time.CLOCK_MONOTONIC_RAW`](https://docs.python.org/3/library/time.html#time.CLOCK_MONOTONIC_RAW) clock).  The command to query that in python is `time.clock_gettime(time.CLOCK_MONOTONIC_RAW)`.  That's subtly different from using `time.monotonic()`; the reference time of the latter is subject to system clock adjustments.  Using the raw monotonic clock we get better multi-process synchronization, but we sacrifice accuracy of the true wall clock time.  I'm inclined to prioritize the former for Redis IDs (which are enforced by Redis to be strictly monotonic anyway), but this tradeoff may be worth discussing in more depth.

- A consequence of the above is that any processes that don't run on the edge will have limited ability to synchronize with edge processes.  Our time-critical code all runs on the edge, so probably this is okay.

- Instead of requiring clients to understand these clock nuances, could we add an `atom.time_monotonic()` function that abstracts this for clients?  It should be simple to provide access to this clock from both Python and C++ versions of `atom`.

- Python implementation proposal: Add a new keyword argument `timestamp` (default is `None`) to [`entry_write`](https://github.com/elementary-robotics/atom/blob/latest/languages/python/atom/element.py#L1138).  `timestamp=None` will maintain current behavior, whereas any other value will be used as the Redis ID with `XADD`.  Note that Redis will throw an error if the requested ID is not monotonically increasing, and we can surface that error to the user -- it's the client's responsibility to ensure the timestamp is valid.

@dpipemazo @mvanveen @MeganParadowski1 curious to hear what you all think after our conversation this morning.  If this all sounds good I can implement this with a short turnaround.

dpipemazo, 2020-09-03:
> That's subtly different from using time.monotonic(); the reference time of the latter is subject to system clock adjustments. 

I didn't know this! Interesting. Coming from C I always used
```
clock_gettime(CLOCK_MONOTINIC)
```
and I thought Python `time.monotonic()` was the same. I should update my metrics PR based off of this info -- thanks.

Overall this seems right to me :+1:

Cody-G, 2020-09-03:
>I didn't know this! Interesting

I didn't pick up on it at first either.  Within a single process `time.monotonic()` is guaranteed to have the same reference time and to be monotonic, it's just that multiple processes aren't guaranteed to have the same reference time.  I think a lot of the new metrics code measures time intervals by calling `monotonic()` twice in a single process -- that should be totally safe.