elementary-robotics/atom #369: Upgrade CircleCI build images

aywaldron, 2020-09-30:
Received the following email from CircleCI:

> Hi there,
> 
> We want to let you know that on October 26th, 2020 we will begin descaling some of the machine images you’re currently using (listed below) . The image will continue to work for you but the average provisioning time will increase.
> 
> For a fast and reliable experience, we suggest updating to one of the supported Linux machine images below before October 26th. This will ensure CircleCI can maintain your image with the speed and support you’ve come to expect.
> 
> The images that will be deprecated no longer receive support from upstream projects. Canonical, the creator of Ubuntu, as well as Docker, no longer manage their images for security or bug fixes. This makes it difficult for us to keep these images operational and secure while maintaining newer images that are supported upstream.
> 
> The deprecations will include:
> 
> Linux machine images:
> circleci/classic:201703-01
> circleci/classic:201708-01
> circleci/classic:201709-01
> circleci/classic:201710-01
> circleci/classic:201710-02
> circleci/classic:201711-01
> circleci/classic:201808-01
> circleci/classic:edge
> circleci/classic:latest
> circleci/classic
> 
> Remote Docker versions:
> 
> docker-17.03.0-ce
> docker-17.05.0-ce
> docker-17.06.0-ce
> docker-17.06.1-ce
> docker-17.07.0-ce
> docker-17.09.0-ce
> docker-17.10.0-ce
> docker-17.11.0-ce
> docker-17.12.0-ce
> docker-17.12.1-ce
> docker-18.01.0-ce
> docker-18.02.0-ce
> docker-18.03.0-ce
> docker-18.03.1-ce
> docker-18.04.0-ce
> docker-18.06.0-ce
> docker-18.09.3
> 
> Suggested images and versions to use instead:
> 
> If you’re using one of the old Linux machine images, we suggest switching to one of these:
> `ubuntu-1604:202007-01` which is an Ubuntu 16.04 based image containing Docker v19.03.12
> `ubuntu-2004:202008-01` which is an Ubuntu 20.04 based image currently in beta.
> 
> For a full list of supported images, refer to CircleCI’s Docs.
> 
> If you’re using a remote Docker image, we suggest specifying a Docker version from v19.03.0 or newer. The list of supported versions can also be found in CircleCI Docs.
> 
> For more details on this change, or if you have any questions, refer to our Discuss post.
> 
> Thanks and happy building,
> 
> The CircleCI team

dpipemazo, 2020-09-30:
Thanks for filing -- most of our jobs are using the `build-ubuntu` executor in our orb which maps to `ubuntu-1604:202004-01`. We needed Docker `19.03` to use `buildx`. I think we should be OK without immediate action on this -- what do you think?

dpipemazo, 2021-04-23:
These were upgraded to the latest recently.

closed 2021-04-23 by dpipemazo