elementary-robotics/atom #375: Behavior of reference_delete when reference

Cody-G, 2020-10-17:
Currently if `reference_delete` gets called on a reference that was already deleted, this error is thrown:

`KeyError: 'Reference reference:hi:46f19324-6ed5-4b12-b706-51151025e6c5:ser:arrow not in redis`

  I'm finding that this complicates the logic for cleaning up references, especially in multiprocess / multithreaded use cases where there can be a race to reference cleanup.  Certainly we could just always write code that avoids this race, but I'm wondering if this strictness really buys us much.  Would there be any harm in allowing deletion of already-deleted references?  I think it's similar in concept to allowing `free` or `del` to be passed a NULL pointer in C / C++.

Here's a link to the current `reference_delete` implementation:

https://github.com/elementary-robotics/atom/blob/latest/languages/python/atom/element.py#L2538-L2556

dpipemazo, 2020-10-17:
@Cody-G we could do this. One argument for strictness is you can simply try/except it in user code to pass on the errors you don't find interesting. I'd prefer to always have the framework be strict and allow the user to explicitly decide to ignore errors. If it dies in the framework then it's never available.

dpipemazo, 2021-04-23:
This has been updated -- `reference_delete` will now return a list of all the references it failed to delete.

closed 2021-04-23 by dpipemazo