elementary-robotics/atom #377: Turn off `E712` in `.flake8` check

dpipemazo, 2020-10-21:
E712 checks that comparisons to booleans are of the `is T/F` type instead of the `== T/F` type. This is dangerous as `is` and `==` do not produce the same outcomes for non-boolean types (numpy ndarray bools, in particular). Blindly applying this rule to code will yield bugs and IMO doesn't add enough to justify the risk.

dpipemazo, 2020-10-21:
Would need to add `E712` to the `ignore =` line in this file: https://github.com/elementary-robotics/atom/blob/latest/utilities/formatting/.flake8

Cody-G, 2020-10-21:
This sounds good to me.  I think this issue summarizes my thoughts pretty well:

https://github.com/PyCQA/pycodestyle/issues/450

TL;DR is that the workarounds for this in numpy are all ugly, so E712 seems too restrictive for numpy programs.

dpipemazo, 2020-10-23:
This was merged in in #378

closed 2020-10-23 by dpipemazo