elementary-robotics/atom #389: Allow for user specification of reference key

aywaldron, 2020-12-10:
When writing a reference, the user should be able to specify the key to use if they want to. This will allow for general Redis key/val storage through atom.

dpipemazo, 2020-12-10:
Awesome! A few questions:

1. Will the user be able to pass the entire key string, i.e. the exact key in redis or will we append the element name onto it? Appending the element name makes scoping a bit easier (less conflicts) but could be more confusing
2. What will we do if the key already exists? How will we ensure no two users use the same key?
3. Should we build it onto the reference API or should we make a new API with a name like "parameters" for this sort of thing?

I don't have strong thoughts/opinions on these, interested to hear what you think.

aywaldron, 2020-12-10:
1. Users will pass the name of the parameter they want to store/get, and we will prefix the key with `parameter:` behind the scenes. This will necessitate a new parameter create/get API separate from the reference API that shares logic for creating and getting the key values. We will not include element names in the keys.
2. We could provide a setting through the parameter API that allows or disallows overrides and is stored in the parameter key. In some cases users will want to write over existing parameters, in some cases they may not want to allow that. We could append `override:true` or `override:false` to the keys and then check for them on `parameter_write`, similarly to how we append the `ser` method to reference keys currently.
3. New parameter API: `parameter_read` and `parameter_write`, as described above.

dpipemazo, 2020-12-10:
@aywaldron for (2) we could actually use a hash field under the hood for this without too much additional complexity and then be able to have an "override" field that we use (set on creation, i.e. key does not exist) as well as a "data" field that we use for the actual parameter. This would be pretty powerful and we could easily add more metadata to parameters using additional hash keys.

Hash Set/Create: https://redis.io/commands/hset
Hash Get Field: https://redis.io/commands/hget
Hash Get all: https://redis.io/commands/hgetall
Hash exists: https://redis.io/commands/hexists
Hash Delete: https://redis.io/commands/hdel

aywaldron, 2020-12-10:
I see, this makes a lot of sense! When it comes to storing objects like the "robot" parameter, are you saying we'd store the multiple fields within the "data" hash field as a dictionary, or as their own hash fields?

dpipemazo, 2020-12-10:
Hm, I think given this we could maybe not imply the "data" hash field and instead have the user supply their own {key : value} dictionary. What do you think?

dpipemazo, 2020-12-10:
Then each value could actually be serialized in any format I guess which is kind of cool. I think we don't really need much more than msgpack, but it could be nice to have the option.

aywaldron, 2020-12-10:
Yeah, it would just require reader elements to know more pieces of information, both the parameter name and the field name. But we can probably manage that through shared specs.

dpipemazo, 2020-12-10:
>  it would just require reader elements to know more pieces of information, both the parameter name and the field name

There's also a "get all" API which could be the primary/default read mode, i.e. return all keys for a parameter. The user could maybe specify exact keys through an optional arg if they want to scope it down.

PR #390 (2020-12-18): Addition of new parameters API
per the discussion in ticket #389, adds new parameters API for storing general key-val pairs in a Redis hash store. 

Also includes:
- Addition of user-specified reference keys
- Bug fix in `reference_delete` method
- Updated docker-compose files to work with most recent atom version
- Addition of submodule cloning to quickstart readme
- Some small documentation udpates/fixes

Resolves #389
closed 2020-12-18 by aywaldron