elementary-robotics/atom #392: Fix example docker-compose.yml file link in docs

slaigai, 2021-01-06:
On the [Atom docs site](https://atomdocs.io/), the link to the example docker-compose.yml file in the [Launching](https://atomdocs.io/?python#launching) section returns a "Failed - Server problem" when attempting to download the file.

dpipemazo, 2021-01-06:
Thanks for the report! Interestingly I can't seem to reproduce; the link points to: http://atomdocs.io/docker-compose/docker-compose.yml which seems to be pointing to a file for me across a few browsers. 

It could be something to do with how we're hosting the site (using one of Heroku's lower tiers), sometimes they don't respond instantly. Are you able to consistently reproduce?

slaigai, 2021-01-06:
Nah, I can't reproduce it today. I was able to consistently reproduce it last night along with a host of other server errors from Heroku (e.g. receiving a Heroku error page when accessing the [Launching](https://atomdocs.io/?python#launching) section directly via URL instead of navigating within the page. Thanks for looking into it! I have the file now.

dpipemazo, 2021-01-06:
Thanks for letting me know! I'll see what I can do Heroku-side.

slaigai, 2021-03-08:
Unable to repro, closing

closed 2021-03-08 by slaigai