elementary-robotics/atom #407: Add type annotations to `languages/python/atom/element.py`

kylebebak, 2021-02-18:
This is actually where the type hint value->effort ratio is highest, because pretty much everything running on the "camera side" depends on Atom.

We don't need all the code in this repo to be annotated, just the method/function signatures, so that anything that depends on Atom can use those type hints (e.g. vision-processing, to which we're adding type hints now). Adding type hints to method/function signatures would require minimal effort.

slaigai, 2021-02-24:
I've added type hints to most of atom to help with the [related ticket](elementary-robotics/vision-processing#126). I'll open a PR for this ticket with changes, but the scope would be expanded to all python files in `languages/python/atom`

slaigai, 2021-03-04:
Expanding scope to add type checker to CI pipeline temporarily with the plan to transition after #413 is sorted out.

PR #408 (2021-03-08): Add type hints to atom python
Changes:
- add type hints to atom (mostly in element.py)
- add type checker (pyright) to test container and include in build checks
- udpate redistimeseries in atom to 1.4.3 using pip to install (stable latest and greatest). This resolves import errors from pyright because redistimeseries used to be a binary built from source. Now pyright can resolve imports correctly.
- update logic to resolve type errors (see changes to tests as well)

Closes #407 , closes #418
closed 2021-03-08 by slaigai