elementary-robotics/atom #421: Document how to install atom and its deps as dev deps for other edge repos

kylebebak, 2021-03-07:
Everything you've done in https://github.com/elementary-robotics/atom/pull/408 means we have type checking on the CI server and in the docker container, but it's **really** important for DX that we get type checking on the host machine, and hence in text editors, as well.

Installing `pyright` on the host machine gets us most of the way there, but it doesn't install deps on the host machine. We can see in [requirements.txt](https://github.com/elementary-robotics/atom/blob/7c82fe3ae56bfc2f9cf6a97b7d5b7305a24731c5/languages/python/requirements.txt) that we've got a bunch of deps for atom, which implicitly become deps for most other edge repos

I think a great place to document this would be in the [Python type checking wiki you made](https://github.com/elementary-robotics/wiki/wiki/Python-Type-Annotations-and-Type-Checking).

dpipemazo, 2021-03-08:
@kylebebak sounds good. Let's document all in the README here.

kylebebak, 2021-03-13:
@slaigai and I synced about this, and I fleshed it out in this wiki page here: https://github.com/elementary-robotics/wiki/wiki/Python-Code-Completion-and-Type-Checking-in-Text-Editors

We're very excited for other members of the team to start using these tools!

dpipemazo, 2021-03-14:
@kylebebak Perhaps we can move this particular doc from our internal Wiki into the wiki  for the open-source Atom project here so that all can see it.

kylebebak, 2021-03-16:
Hey @dpipemazo , I like this idea. There are a few things I'd have to add to the current wiki so it's ready for "external" consumption, but once I add those things I'll make it public