elementary-robotics/atom #427: Add `read` for new queues

dpipemazo, 2021-03-12:
Once #404 is merged, there's one API that would be great to have: the ability to read from the queue without consuming (introspection). This is a huge advantage to the redis sorted set over the Python Multiprocessing Queue.

It should be fairly straightforward to implement using a `ZRANGE` or similar command and an API similar to `get_n`.

slaigai, 2021-03-12:
I agree, this would have been useful in one of my other PRs. 

I think we could call it "peek". Like "look but don't touch". Personally, I agree that "read" is OK because it makes me think of file readers that consume lines as they parse files. I associate "inspect" with inspection of the queue's properties rather than its items but I might be in the minority with this viewpoint (maybe).

dpipemazo, 2021-03-12:
`peek` sounds great to me!

PR #436 (2021-03-16): add peek to queues
Changes:
- Add peek functionality to read at most N items from queues without consuming items.
- Add tests for peek
- Bugfix: AtomFIFOQueue with `max_highest_prio=True` did not return items in FIFO order. This argument is now silently ignored to maintain compatibility with the AtomPRIOQueue interface.

Closes #427
closed 2021-03-16 by slaigai