elementary-robotics/atom #432: RFC: Add CD to `PyPI` on tagged releases

dpipemazo, 2021-03-13:
Should we add deployment to PyPI as part of our tagged release process?

slaigai, 2021-03-17:
After going through [this issue](https://github.com/elementary-robotics/atom/issues/421) and testing out the workflow, I personally would love to have deployment as an automated part of the tagged release process. My current workflow involves pulling latest (atom) and `python setup_local.py install`ing and sometimes I install atom from a working branch by mistake. It would be much easier to `pip install --upgrade atom` :).

dpipemazo, 2021-03-17:
It should be fairly straightforward to do off of tagged release builds. @kylebebak what do you think?

kylebebak, 2021-03-17:
Are we talking about automatically deploying to PyPI whenever we create a new tag for atom? I think that would be great.

It might also be nice to simplify bumping the atom version. Currently we change version strings in 4 files:

- `docker-compose-dev.yml`
- `languages/python/setup.py`
- `languages/python/setup_local.py`
- `languages/python/atom/config.py`

Maybe changing these versions manually is the easiest, most reliable way to do this. But maybe there's a better way, and maybe it's not necessary to change it in all 4 places (e.g. could we just the `BASE_IMAGE` point to the `latest` tag in `docker-compose-dev.yml`?)

kylebebak, 2021-03-19:
@slaigai found this link that goes through exactly how to do this using Circle!

https://circleci.com/blog/publishing-a-python-package/